ARM-software/sysarch-acs #13: 【BSA-Test】About linux bsa test-862“No extra address translation” (Ported from bsa-acs)

rajatgoyal47, 2025-06-26:
Original Issue Link : https://github.com/ARM-software/bsa-acs/issues/487

about DMA addr,as " *dma_addr = bsa_dma_addr; *dma_len = bsa_dma_len;"
We are using the address that has been applied for by the driver, but we cannot control when the driver is released.This led to the occasional failure of our 862 test.

_862: No extra address translation
Device IOVA entry is fffff000 size = 1000 phys = 1803d0000
Device IOVA entry is fffc0000 size = 20000 phys = 10009e6a4000
Device IOVA entry is fffa0000 size = 20000 phys = 10009e6bc000
Device IOVA entry is fff80000 size = 20000 phys = 10009e6d4000
Device IOVA entry is fff60000 size = 20000 phys = 10009e6fc000
Device IOVA entry is ffcdf000 size = 1000 phys = 1000affbc000

The DMA address ffce0000 used by device
is not present in the SMMU IOVA table

Failed on PE - 0
Checkpoint -- 0 : Result: FAIL_

How can this problem be avoided and is there a good solution

thanks

Attaching the log from issue here : 
[Bsalog.txt](https://github.com/user-attachments/files/20917603/Bsalog.txt)

CC : @JackWeihuadi

rajatgoyal47, 2025-06-30:
Hi @JackWeihuadi,
 
We’ve updated the Linux application build process. Please refer to the instructions here: https://github.com/ARM-software/sysarch-acs/?tab=readme-ov-file#linux-application.
 
This updated guide allows you to build the Linux application directly on the target machine using build.sh, without needing to apply any patches beforehand. It covers the steps required to compile the BSA/SBSA application binaries.
 
Can you please share your findings?
 
Regards,
ACS Team

JackWeihuadi, 2025-07-01:
Hi @rajatgoyal47 

  we get from https://github.com/ARM-software/arm-systemready/tree/v25.04_SR_3.0.1/SR/prebuilt_images

rajatgoyal47, 2025-07-14:
Hi @JackWeihuadi,

Can you please use the image available at : https://github.com/ARM-software/arm-systemready/actions/runs/16251781359 
This has latest changes from the updated build process.

Please share the logs with us.

Regards,
ACS Team

chetan-rathore, 2025-12-04:
Hi @JackWeihuadi, 

As the sysarch-acs based ACS were going rule based changes, there was delay in follow up of the issue.
We will take this on priority and try to share updated by next week.

chetan-rathore, 2026-02-20:
@JackWeihuadi, Did you get chance to run the latest BSA/SBSA linux application