ARM-software/sysarch-acs #42: Multiple build failures in ACS BSA/SBSA kernel modules

shakum06, 2025-07-21:
I encountered multiple build issues while using the ACS `build.sh` script on an AArch64 platform (native build with kernel 6.10). The build fails at several stages for both BSA and SBSA kernel modules due to missing license macros and unresolved symbols.

---

## ❗ Key Problems

ERROR: modpost: "g_print_level" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/platform/pal_linux/files/bsa_acs_pal.ko] undefined!

ERROR: modpost: missing MODULE_LICENSE() in /home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.o
ERROR: modpost: "pal_smmu_device_start_monitor_iova" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "pal_mem_virt_to_phys" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "pal_mem_compare" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "pal_mem_free_pages" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "g_acs_tests_pass" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "pal_pe_get_far" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "g_bsa_only_level" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "pal_is_bdf_exerciser" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "g_num_tests" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
ERROR: modpost: "pal_pe_update_elr" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/val/bsa_acs_val.ko] undefined!
WARNING: modpost: suppressed 97 unresolved symbol warnings because there were too many)

ERROR: modpost: missing MODULE_LICENSE() in /home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.o
ERROR: modpost: "val_pcie_device_driver_present" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_memory_free" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_pcie_get_info" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_check_for_error" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_pe_get_mpid" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_pcie_is_devicedma_64bit" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_pe_get_index_mpid" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_dma_device_get_dma_addr" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_print" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
ERROR: modpost: "val_initialize_test" [/home/shakum06/acs/linux-acs/acs-drv/files/linux-acs/acs-drv/files/test_pool/bsa_acs_test.ko] undefined!
WARNING: modpost: suppressed 28 unresolved symbol warnings because there were too many)

rajatgoyal47, 2025-07-22:
Hi @shakum06,

These errors appear because the val, pal, and test_pool layers are compiled separately. Once all modules are combined, the bsa/sbsa_acs.ko file is generated, so you can ignore these intermediate errors.

Please verify if the .ko files are present in the build folder. Let us know if you need any further information.

Regards,
ACS Team

rajatgoyal47, 2025-07-30:
Hi @shakum06,

Please let us know if you need any further assistance. Otherwise, we’ll go ahead and close this.

Regards,
ACS Team

shakum06, 2025-07-30:
.ko file was generated from build.sh script, ran some test also. We can close the issue.