ARM-software/sysarch-acs #49: BSA ACS is not buildable with GCC 15

hrw, 2025-07-27:
My Fedora 42 box uses gcc 15:

```
$ gcc --version
gcc (GCC) 15.1.1 20250521 (Red Hat 15.1.1-2)
```

Build fails:

```
Building ... /home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Library/UefiApplicationEntryPoint/UefiApplicationEntryPoint.inf [AARCH64]
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/src/pal_gic.c: In function ‘pal_gic_install_isr’:
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/src/pal_gic.c:217:69: error: passing argument 3 of ‘gInterrupt->RegisterInterruptSource’ from incompatible pointer type [-Wincompatible-pointer-types]
  217 |   Status = gInterrupt->RegisterInterruptSource (gInterrupt, int_id, isr);
      |                                                                     ^~~
      |                                                                     |
      |                                                                     void (*)(void)
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/src/pal_gic.c:217:69: note: expected ‘HARDWARE_INTERRUPT_HANDLER’ {aka ‘void (*)(long long unsigned int,  EFI_SYSTEM_CONTEXT)’} but argument is of type ‘void (*)(void)’
In file included from /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/src/pal_gic.c:25:
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/EmbeddedPkg/Include/Protocol/HardwareInterrupt.h:47:10: note: ‘HARDWARE_INTERRUPT_HANDLER’ declared here
   47 | (EFIAPI *HARDWARE_INTERRUPT_HANDLER)(
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/src/pal_gic.c:220:71: error: passing argument 3 of ‘gInterrupt->RegisterInterruptSource’ from incompatible pointer type [-Wincompatible-pointer-types]
  220 |     Status = gInterrupt->RegisterInterruptSource (gInterrupt, int_id, isr);  //register our Handler.
      |                                                                       ^~~
      |                                                                       |
      |                                                                       void (*)(void)
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/src/pal_gic.c:220:71: note: expected ‘HARDWARE_INTERRUPT_HANDLER’ {aka ‘void (*)(long long unsigned int,  EFI_SYSTEM_CONTEXT)’} but argument is of type ‘void (*)(void)’
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/EmbeddedPkg/Include/Protocol/HardwareInterrupt.h:47:10: note: ‘HARDWARE_INTERRUPT_HANDLER’ declared here
   47 | (EFIAPI *HARDWARE_INTERRUPT_HANDLER)(
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: Wejście do katalogu '/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/MdePkg/Library/UefiApplicationEntryPoint/UefiApplicationEntryPoint'
make[1]: Nie ma nic do zrobienia w 'tbuild'.
make[1]: Opuszczenie katalogu '/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/MdePkg/Library/UefiApplicationEntryPoint/UefiApplicationEntryPoint'
make[1]: *** [GNUmakefile:414: /home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/UnifiedPalLib/OUTPUT/src/pal_gic.obj] Błąd 1
make[1]: Opuszczenie katalogu '/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/UnifiedPalLib'
Building ... /home/marcin/devel/linaro/sbsa-qemu/code/edk2/MdePkg/Library/RegisterFilterLibNull/RegisterFilterLibNull.inf [AARCH64]

build.py...
 : error 7000: Failed to execute command
        make tbuild [/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/sysarch-acs/pal/uefi_acpi/UnifiedPalLib]
```

This does not happen with:
- GCC 13.2 used by Github Actions 
- GCC 12.2 present in Debian 'bookworm' (current stable)
- GCC 14.2 present in Debian 'trixie' (soon to be stable)

rajatgoyal47, 2025-07-28:
Hi @hrw,

Could you kindly review and verify the following changes?
https://github.com/rajatgoyal47/sysarch-acs/commit/23ecb2830d0913b93e921ddd75ee3caf167f8220

Please share your feedback. 

Best regards,
ACS Team

hrw, 2025-07-28:
Yes, that change sorts out the problem.