ARM-software/sysarch-acs #105: BSA Test 904: Test Fails even when MSI-X is initiated

tanwar1981, 2025-09-10:
Environment: QEMU based environment where PCIe Avery VIP is acting as exerciser (System Verilog)
We are running BSA Test 904.

**First Issue:** Test rightly enable the "MSI-X Enable" bit in PCIe config space but doesn't clear the Mask bit in "Vector Control Register for MSI-X Table Entries". Default value of mask bit is 1 as per PCIe spec and hence needs to be cleared by host

**Second issue:** When test enables exerciser for MSI-X initiation by writing 1 to MSICTLTRG (Bit 0 of BSA MSI Control Register), it doesn't care to set right index of MSI-X. Whatever comes in the completion of Memory read it just write same value. See the value "eeeeeeee"

==> @222566.391ns  MRD#a872 (addr 81100004, 1 dwords)(req_id 0, tag 1c)
    | fmt |  type  |t9|  tc  |t8|a2|r|th|td|ep|att| at|    length   |
    |__0__|___00___|_0|___0__|_0|_0|0|_0|_0|_0|_0_|_0_|_____001_____|
    |________ req_id: 0000__________|___ tag: 1c____|_lbe:0_|_fbe:f_|
    |_____________________ addr[31:2]: 20440001_______________|ph: 0|
    00000001 00001c0f 81100004

                <== @222566.393ns  CPLD#a875 (MRD#a872, SUCCESS)(req_id 0, tag 1c) (MRD#a872)
                    | fmt |  type  |t9|  tc  |t8|a2|r|th|td|ep|att| at|    length   |
                    |__2__|___0a___|_0|___0__|_0|_0|0|_0|_0|_0|_0_|_0_|_____001_____|
                    |___ cpl_id: 0100 ____|cpl_status: 0_|bcm: 0|__ byte_cnt: 004 __|
                    |___ req_id: 0000 ____|___ tag: 1c __|rsvd20: 0|__low_addr: 04__|
                    4a000001 01000004 00001c04
                    **eeeeeeee**

==> @222730.391ns  MWR#a882 (addr 81100004, 1 dwords)
    | fmt |  type  |t9|  tc  |t8|a2|r|th|td|ep|att| at|    length   |
    |__2__|___00___|_0|___0__|_0|_0|0|_0|_0|_0|_0_|_0_|_____001_____|
    |________ req_id: 0000__________|___ tag: c3____|_lbe:0_|_fbe:f_|
    |_____________________ addr[31:2]: 20440001_______________|ph: 0|
    40000001 0000c30f 81100004
    eeeeeeee

**Third Issue:**  After above sequence exerciser initiates one MSI-X. But it picks address and data from default value in MSI-X Vector Control register. Looks like Test 904 not fully exercising MSI-X table. After going through the test 904 in detail I found following code which actually configures the MSI-X table

/* Fill MSI Table with msi_addr, msi_data */
  val_mmio_write(table_address + msi_index*MSI_X_ENTRY_SIZE + MSI_X_MSG_TBL_LOWER_ADDR_OFFSET,
                                                                          (uint32_t)msi_addr);
  val_mmio_write(table_address + msi_index*MSI_X_ENTRY_SIZE + MSI_X_MSG_TBL_HIGHER_ADDR_OFFSET,
                                                     (uint32_t)(msi_addr >> MSI_X_ADDR_SHIFT));
  val_mmio_write(table_address + msi_index*MSI_X_ENTRY_SIZE + MSI_X_MSG_TBL_DATA_OFFSET, msi_data);
  val_mmio_write(table_address + msi_index*MSI_X_ENTRY_SIZE + MSI_X_MSG_TBL_MVC_OFFSET, 0x0);

But this code doesn't configure the exerciser's table. So exerciser is not aware of configured address and data for MSI-X

Not sure if this is the reason why this test fails even after host receives MSI-X interrupt

<== @222730.393ns  MWR#a886 (addr 44091040, 1 dwords) (APCI_TRANS_mem#a885)
                    | fmt |  type  |t9|  tc  |t8|a2|r|th|td|ep|att| at|    length   |
                    |__2__|___00___|_0|___0__|_0|_0|0|_0|_0|_0|_0_|_0_|_____001_____|
                    |________ req_id: 0100__________|___ tag: 71____|_lbe:0_|_fbe:f_|
                    |_____________________ addr[31:2]: 11024410_______________|ph: 0|
                    40000001 0100710f 44091040
                    0000004c
                    ((MSIx request: addr 44091040, vec 6ee))

Following message appears from QEMU log
===========================================================
QEMU] pcie_config_read: bdf: 01:00:0 addr: 38 data: 0 size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 3c data: 1ff size: 4 bytes
 ECAM base address is F0000000
[QEMU] pcie_mmio_cmd: read: addr: 81100004 data: eeeeeeee size: 4 bytes
[QEMU] rp_cmd_rw: endpoint write address: 44091040
[QEMU] addr: 44091040:  4c 00 00 00                                        L...
[QEMU] pcie_mmio_cmd: write: addr: 81100004 data: eeeeeeee size: 4 bytes

       Interrupt trigger failed for : 0x204C, BDF : 0x10000   [QEMU] pcie_config_read: bdf: 01:00:0 addr: 34 data: 5c size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 5c data: 38036401 size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 64 data: 808005 size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 80 data: 87ffb411 size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 80 data: 87ffb411 size: 4 bytes
[QEMU] pcie_config_write: bdf: 01:00:0 addr: 80 data: 7ffb411 size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 84 data: 0 size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 10 data: 8100000c size: 4 bytes
[QEMU] pcie_config_read: bdf: 01:00:0 addr: 14 data: 0 size: 4 bytes
[QEMU] pcie_mmio_cmd: write: addr: 81000000 data: 00000000 size: 4 bytes
[QEMU] pcie_mmio_cmd: write: addr: 81000004 data: 00000000 size: 4 bytes
[QEMU] pcie_mmio_cmd: write: addr: 81000008 data: 00000000 size: 4 bytes
[QEMU] pcie_mmio_cmd: write: addr: 8100000c data: 00000001 size: 4 bytes

       Failed on PE -    0
       PCI_MSI_2,ITS_DEV_6
       Checkpoint --  6                           : Result:  FAIL
       END

      One or more Exerciser tests failed or were skipped.

     -------------------------------------------------------
     Total Tests run  =    1  Tests Passed  =    0  Tests Failed =    1
     -------------------------------------------------------

      *** BSA tests complete. Reset the system. ***

SrikarJosyula, 2025-09-16:
Hi @tanwar1981,

Thanks for sharing the details. 
Just a quick note - the APIs in [pal_exerciser.c](https://github.com/ARM-software/sysarch-acs/blob/main/pal/baremetal/target/RDN2/src/pal_exerciser.c) are written for a specific exerciser reference design as described in [Exerciser.md](https://github.com/ARM-software/sysarch-acs/blob/main/docs/pcie/Exerciser.md). On user's platform, these APIs may need to be updated or adjusted to match the behavior of the target exerciser implementation.

On the issues you raised,

- In the test infra, the mask bit in Vector Control Register for each MSI-X entry is set to 0 (unmasked). You can see this in the code here [acs_gic_support.c#L407](https://github.com/ARM-software/sysarch-acs/blob/main/val/src/acs_gic_support.c#L407)

- The MSI-X vector index is passed to the exerciser through a PAL API ([pal_exerciser.c#L551](https://github.com/ARM-software/sysarch-acs/blob/main/pal/baremetal/target/RDN2/src/pal_exerciser.c#L551)). This index is what the exerciser uses to locate the correct MSI-X address and data in order to trigger the interrupt.

To help narrow things down further, a few checks might be useful:

- ⁠Please confirm that the PAL exerciser APIs in your setup are aligned with your platform design. The file to update is [pal_exerciser.c](https://github.com/ARM-software/sysarch-acs/blob/main/pal/baremetal/target/RDN2/src/pal_exerciser.c). For more info please refer [Exerciser API Porting Guide](https://github.com/ARM-software/sysarch-acs/blob/main/docs/pcie/Exerciser_API_porting_guide.md).

- ⁠Verify that the "Function Mask" bit in the MSI-X Message Control register is not set. ACS doesn’t change this bit (it defaults to 0), but if it’s set to 1 in your environment, then all MSI-X vectors will be masked.

- ⁠Check that the GIC ITS is fully configured and initialized so that it can receive and handle MSI transactions correctly.

If you still see the issue after the above, kindly provide more logs so we can investigate in more detail.

Thanks,
Srikar

tanwar1981, 2025-09-26:
Hi Srikar,
I can see MSI-X table getting updated and mask bit is also getting clear but still test fails.

After updating table test reads the MSI-X Control register (BSA Register) at offset 81100004
===========================================================
[QEMU] pcie_mmio_cmd: read: addr: 81100004 data: eeeeeeee size: 4 bytes
============================================================

Here you can see data returned from EP exerciser is eeeeeeee. Next test writes the same data by enabling MSICTLTRG  (Bit 31)
==============================================================
[QEMU] pcie_mmio_cmd: write: addr: 81100004 data: eeeeeeee size: 4 bytes
==============================================================

Here MSICTLID (Bit 10:0) remains 6ee (lower 11 bits from eeeeeeee). This means vector value is configured for 6ee (1774 in decimal)

Now exerciser initiates MSI-X on configured address and data with vector 6ee

AVY_INF: apci_test_log@238606.391ns: Host enabled MSI from Endpoint
11897 ep0@238606.391ns  D0_F0 do:  APCI_TRANS_mem#97a2 addr 44091040, length 1, is_write 1, first_be f, last_be f (user_ctrl: th 3), payload: 0000004c  (MSIx request: addr 44091040, vec 6ee)
11898 
11899                 <== @238606.393ns  MWR#97a3 (addr 44091040, 1 dwords) (APCI_TRANS_mem#97a2)
11900                     | fmt |  type  |t9|  tc  |t8|a2|r|th|td|ep|att| at|    length   |
11901                     |__2__|___00___|_0|___0__|_0|_0|0|_0|_0|_0|_0_|_0_|_____001_____|
11902                     |________ req_id: 0100__________|___ tag: 02____|_lbe:0_|_fbe:f_|
11903                     |_____________________ addr[31:2]: 11024410_______________|ph: 0|
11904                     40000001 0100020f 44091040
11905                     0000004c
11906                     ((MSIx request: addr 44091040, vec 6ee))

But still test fails.

[QEMU] pcie_mmio_cmd: write: addr: 8100000c data: 00000001 size: 4 bytes
^M
       Failed on PE -    0^M
       PCI_MSI_2,ITS_DEV_6^M
       Checkpoint --  6                           : Result:  FAIL^M
       END^M
^M
^M
      One or more Exerciser tests failed or were skipped.^M
^M
     -------------------------------------------------------^M
     Total Tests run  =    1  Tests Passed  =    0  Tests Failed =    1^M
     -------------------------------------------------------^M
^M
      *** BSA tests complete. Reset the system. ***^M
^M
^[[1m^[[33m^[[40mFS0:\> ^[[0m^[[37m^[[40mERROR: SimCluster disconnected!
qemu-system-aarch64: /machine/peripheral/avy: Disconnected clk=4271007646981 ns

Let me know what you need from my side. I am not sure why test fails but looks like correct vector value may be the problem.